oliverbravery/PrintGuard #16: Home Assistant Support

oliverbravery, 2025-07-10:
Add support for home assistant, so that printers can link using home assistant api keys for example.

paulbaird87, 2025-08-14:
YES Pleeeeeaaaase

Would make this, already excellent tool, infinitely more versatile.

bnevets27, 2025-08-14:
> YES Pleeeeeaaaase
> 
> Would make this, already excellent tool, infinitely more versatile.

I suggest reacting to the issue. Usually better than a lot of +1s. But I agree fully.

This would be critical for me as HA is the only way I can interact with my printer.

louispires, 2026-01-02:
> > I am checking out the `webui` branch and building the container locally - Is there a chance that it is functional at all [@oliverbravery](https://github.com/oliverbravery) ?
> 
> Hi, that branch is partially functional. Webcams and HomeAssistant components (via connections NOT the integration) are working alongside detection. There are a lot of issues however as this was more of a prototype. I am developing a more complete solution but do not currently have an eta on when that will be released.

I managed to check out the code, fix a couple of bugs and got the RTSP stream and Inferencing working:

[image]

Seems like notifications don't work on Edge but Chrome is fine.