oliverbravery/PrintGuard #36: Feature Request - Prusa Connect

reuter, 2025-07-20:
One of the things I miss most switching from Octoprint was being able to use Obico.
Always had the idea to do the Spaghetti detection standalone and then send over commands.
So.. BIG thank you for doing this!!

Great feature would be able to link this up to Prusa Connect printers.
https://github.com/prusa3d/Prusa-Connect-SDK-Printer

emptyturtle, 2025-07-21:
Hey I tried to make something like that in my fork, but I used the PrusaLinkPy API that connects through PrusaLink. I tried it myself and with my Core One it worked but it is not finished but it can pause if it detects a failure. Maybe would this Prusa Connect SDK be better for that but I am not sure. If you want you can try it and give me your feedback.

emptyturtle, 2025-08-22:
> Works amazing.
> I did find that it was pausing the print before it even got started. But I don't think that was the prusa implementation doing that.
> thank you very much for this!

Thank you! Really happy to hear this feedback. Just message me if you have a problem with it.